brainelectronics/micropython-modbus #11: Lib only allows one TCP connection at a time

brainelectronics, 2022-07-16:
As found out during #10 only one TCP Modbus connection can be established. If a second client connects to the master device, the first connection will be closed.

The parameter `max_connections` of the TCPServer class in function `bind` is misleading

see https://github.com/brainelectronics/micropython-modbus/blob/dd5b8e593697d098b29abf2615d57216c7346f4a/umodbus/tcp.py#L221

brainelectronics, 2022-10-23:
related to #18

beyonlo, 2023-01-15:
Hello @brainelectronics are you thinking to solve this bug as well?

brainelectronics, 2023-01-16:
Good morning @beyonlo yes I'm planning to solve all of these issues. To be honest I have to rearrange prio a bit as I was exclusively working for this lib the last few weeks. Nevertheless I really try to get this bug as well as the strange CRC and Serial Control pin bug done. After that I'll quickly do the async milestone

beyonlo, 2023-01-16:
Hello @brainelectronics good morning :)

> Good morning @beyonlo yes I'm planning to solve all of these issues.

That's great

>To be honest I have to rearrange prio a bit as I was exclusively working for this lib the last few weeks.

I really realized that you was very very active in this project. You put this project in another level in the lasts weeks, in implementations and docs! Thank you so much for your effort!

> Nevertheless I really try to get this bug as well as the strange CRC and Serial Control pin bug done. 

Excellent!

>After that I'll quickly do the async milestone

Amazing, this feature will be very useful!  :partying_face: