brainelectronics/micropython-modbus #12: Coil quantity specification does not work

brainelectronics, 2022-07-16:
As found out during #10 the `coil_qty` parameter of the `read_coils` function of [TCP](https://github.com/brainelectronics/micropython-modbus/blob/dd5b8e593697d098b29abf2615d57216c7346f4a/umodbus/tcp.py#L94) and [RTU](https://github.com/brainelectronics/micropython-modbus/blob/dd5b8e593697d098b29abf2615d57216c7346f4a/umodbus/serial.py#L196) is not working. 

```python
host.read_coils(slave_addr=10, starting_addr=123, coil_qty=1)
# [True, False, False, False, False, False, False, False]   # <--- "coil_qty=1" but showed 8 and not just 1.
host.read_coils(slave_addr=10, starting_addr=123, coil_qty=20)
# [True, False, False, False, False, False, False, False]   # <--- "coil_qty=20" but still showing 8, and not 20.
```

javiercp64, 2022-07-20:
And only the first coil value is correct, the others  always appears as false.

beyonlo, 2022-07-22:
> And only the first coil value is correct, the others always appears as false.

That's correct @javiercp64 for [default definitions example](https://github.com/brainelectronics/micropython-modbus/blob/bugfix/improve-docs-and-add-examples/examples/rtu_client_example.py#L36) because the len is `1` (`"len": 1`) for the value `1` (`"val": 1`), so others bits that are not set, are `0` (`False`) by default.

If you change the definitions to set others bit values you will receive `True` if you set as `1`. See example below with new definitions - look at `"val"`:

`register_definitions = {    "COILS": {        "EXAMPLE_COIL": {            "register": 123,            "len": 3,            "val": [1, 0, 1]     }   },
`

**Output:**
```
>>> host.read_coils(slave_addr=10, starting_addr=123, coil_qty=3)
[True, False, True, False, False, False, False, False]
>>>
```

However, the write multiple COILS has a bug, as reported on the PR #10, in [this section](https://github.com/brainelectronics/micropython-modbus/pull/10#issuecomment-1189078241), `item 2`.

beyonlo, 2022-09-02:
As detected by @brainelectronics on the thread https://github.com/brainelectronics/micropython-modbus/pull/10#issuecomment-1203521346 is correct to return always multiple of 8 bits. So is correct when `coil_qty=1` returning the 8 bits/elements. But the bug still exists, once that `coil_qty=20` is returning just first 8 bits.

```Python
host.read_coils(slave_addr=10, starting_addr=123, coil_qty=1)
[True, False, False, False, False, False, False, False]   # CORRECT: "coil_qty=1"  and showed 8 bits (multiple of 8).
host.read_coils(slave_addr=10, starting_addr=123, coil_qty=20)
[True, False, False, False, False, False, False, False]   # NOT CORRECT: "coil_qty=20" showing only first 8 bits.
```

brainelectronics, 2022-09-04:
@brainelectronics 
Check Implementation to follow requirement of returning multiple coils as requested, see http://www.simplymodbus.ca/FC01.htm

brainelectronics, 2022-11-06:
https://github.com/brainelectronics/micropython-modbus/compare/develop...ShaunL36:micropython-modbus:develop

brainelectronics, 2022-11-12:
related to #25

brainelectronics, 2022-12-02:
https://github.com/brainelectronics/micropython-modbus/commit/220d3b9ea84cb5c913168e64dddd7f2c6ab9e7c5

closed 2022-12-11 by brainelectronics