brainelectronics/micropython-modbus #15: Question: how to use the write_single_coil() to write on the different positions?

beyonlo, 2022-09-01:
Hello @brainelectronics 

I'm using the https://github.com/brainelectronics/micropython-modbus/pull/10 for tests.

How is possible to use the `write_single_coil()` to write in position 3, for example? In this example below the `ouput_value` is setting always to the first position:

```
>>> host.read_coils(slave_addr=10, starting_addr=123, coil_qty=1)
[False, True, True, False, False, False, False, False]
>>> host.write_single_coil(slave_addr=10, output_address=123, output_value=1)
False
>>> host.read_coils(slave_addr=10, starting_addr=123, coil_qty=1)
[True, False, False, False, False, False, False, False]
>>> 
```

Using this register_definitions: 
`{'ISTS': {'EXAMPLE_ISTS': {'register': 67, 'len': 1, 'val': 0}}, 'IREGS': {'EXAMPLE_IREG': {'register': 10, 'len': 2, 'val': 60001}}, 'HREGS': {'EXAMPLE_HREG': {'register': 93, 'len': 1, 'val': 19}}, 'COILS': {'EXAMPLE_COIL': {'register': 123, 'len': 3, 'val': [0, 1, 1]}}}
`

beyonlo, 2022-11-11:
**Very importat:** I suppose that when use `output_value` without a `list`, I mean, just `1/0/True/False` like as example above, it will be write on the first position, (position `0`), is that right? That is a bit confuse. In my opinion is more clear always to use a list when need to write COILS. So even that you need to write just one COIL, in the first position, need to use always `output_value=[1]`, What do you think? That will prevents confusion. So in this case the register definitions need always to use list as well: `EXAMPLE_COIL': {'register': 123, 'len': 3, 'val': [0,1,1]` or if has all 0/False, can use empty list: `EXAMPLE_COIL': {'register': 123, 'len': 3, 'val': [ ]` . Anyway, always is need to be documented: as each register is by definition 2 bytes (16 bits), so coils that are not defined, is by default `0/False`. So `EXAMPLE_COIL': {'register': 123, 'len': 3, 'val': [0,1,1]` is the same thing that to do `EXAMPLE_COIL': {'register': 123, 'len': 3, 'val': [0,1,1,0,0,0,0,0,0,0,0,0,0,0,0,0]`. Is that make sense for you?

brainelectronics, 2022-12-08:
Relates to #26

brainelectronics, 2022-12-16:
Relates to #35, same root cause for this issue. Access to the elements of a list of a register value is missing/not yet possible

> Registers are defined as specified by the `register` element, see https://github.com/brainelectronics/micropython-modbus/blob/c45d6cc334b4adf0e0ffd9152c8f08724e1902d9/umodbus/modbus.py#L53-L55 for a coil. The provided `value` is added as value to the dictionary at the key `register`, here `93`. Thereby no element/key `94` exists in the Modbus `_register_dict`

PR #53 (2023-01-08): Allow client register access and setting at any position
### Added
- Unittest to read multiple coils at any location if defined as list, verifies #35
- Unittests to write a single coil or multiple coils at any location if defined as list, verifies fix #15 and #24

### Fixed
- All configured register of a client can be accessed and modified individually, see #35
- Resolved overlapping register positions in example JSON file
- Register length of `EXAMPLE_IREG` in TCP and RTU examples corrected to 1 instead of 2
closed 2023-01-08 by brainelectronics