brainelectronics/micropython-modbus #16: Create test framework

brainelectronics, 2022-09-04:
As found during #10 the implementation does not 100% follow the Modbus requirements 

Create test framework with different requests and check response data for single and multi registers

brainelectronics, 2022-11-30:
@beyonlo test framework is almost done. With this corresponding Branch. Confident to provide a fix on the upcoming weekend. Unfortunately there is a breaking change. But it only about the way ModbusRTU and ModbusTCP classes are imported.

beyonlo, 2022-11-30:
That's a great news @brainelectronics  :)

I think that you do not need to worry about breaking backwards compatibility for now, because this is a lib that is still new and growing, so, in my opinion new features and bug fixes are more important than maintaining compatibility.

beyonlo, 2022-11-30:
> You're right. The only issue are my other packages which are badend on this Lib and the user can just update. My used version of upip does not allow fixing versions of dependency packages so I have to release a new version of the "parent" package on every breaking release. No problem but unpleasant.

Well, If you can at least to automate the process to generate new releases of dependency packages, maybe can help you!

> BTW #21 has been fixed partying_face with the next release

Excellent! :tada:  :tada: :tada:

closed 2022-12-11 by brainelectronics