brainelectronics/micropython-modbus #17: ModbusRTU Client

osorioen, 2022-10-08:
# RTU Slave setup
             # act as client, provide Modbus data via RTU to a host device
              rtu_pins = ('P3', 'P4')         # (TX, RX)
        
                slave_addr = 10             # address on bus as client
                  baudrate = 9600
line 679        client = ModbusRTU(addr=slave_addr , baudrate=9600,pins= rtu_pins)

File "/flash/lib/umodbus/modbus.py", line 679, in __init__
  File "/flash/lib/umodbus/serial.py", line 42, in __init__
TypeError: extra keyword arguments given

beyonlo, 2022-10-10:
@osorioen I'm not sure, but I think that you need to pass to the `rtu_pins` argument a real PIN number tuple, like as `rtu_pins = (25, 26)`. I believe that you are using a `STM` port, that use names `P3,` `P4,` `X3,` etc instead a number pin. I see that https://github.com/brainelectronics/micropython-modbus/blob/develop/umodbus/serial.py#L12 use the `machine.UART()` that following the [docs](https://docs.micropython.org/en/latest/library/machine.UART.html?highlight=uart#machine.UART.init) the `tx` and `rx` wait for a `pin number`. I have no a `stm` Microcontroller here to test, but I think that if you set the `rtu_pins` as pin numbers will works.

jgcarlson, 2022-11-12:
@osorioen not sure what microcontroller you're using, but the suggestion from @beyonlo worked in my case.. Using [`machine.Pin(int)`](https://docs.micropython.org/en/latest/library/machine.Pin.html#class-pin-control-i-o-pins) with the GP number solved the problem for me.

brainelectronics, 2022-11-13:
As suggested Using [`machine.Pin(int)`](https://docs.micropython.org/en/latest/library/machine.Pin.html#class-pin-control-i-o-pins) with the GP number will solve the problem.

brainelectronics, 2022-12-16:
Hi @osorioen,
is your problem resolved with the answer provided by @beyonlo and the now available documentation at [RTD MicroPython Modbus](https://micropython-modbus.readthedocs.io/en/latest/)?

brainelectronics, 2022-12-31:
Resolved by #45

closed 2022-12-31 by brainelectronics