brainelectronics/micropython-modbus #21: Bug: incorrect return when write a COIL with 1/True

beyonlo, 2022-11-11:
This bug was extracted from PR #10 report, in this [reply](https://github.com/brainelectronics/micropython-modbus/pull/10#issuecomment-1189078241):

On the `host.write_single_coil()` has incorrect return when write 1/True.

I believe that return `(True` or `False)` on write methods is to check if has success or not writing data, is this interpretation correct?
When I write `COIL` value as `1` or `True,` it works, I mean, that value is writing with success, but the return is `False` (not success), differently when I write `0` or `False,` where the return is `True` (success).

```
>>> host.write_single_coil(slave_addr=10, output_address=123, output_value=0)
True
>>> host.read_coils(slave_addr=10, starting_addr=123, coil_qty=1)
[False, False, False, False, False, False, False, False]
>>> host.write_single_coil(slave_addr=10, output_address=123, output_value=1)
False
>>> host.read_coils(slave_addr=10, starting_addr=123, coil_qty=1)
[True, False, False, False, False, False, False, False]
>>> 
```

brainelectronics, 2022-11-12:
See [Modbus Application Protocol](https://modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf) section 6.5

> The normal response is an echo of the request, returned after the coil state has been written.

beyonlo, 2022-11-12:
> See [Modbus Application Protocol](https://modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf) section 6.5
> 
> > The normal response is an echo of the request, returned after the coil state has been written.

All right. Understood. So, the response is not the result if was wrote with success or not. The response is just the a echo of the same value that was wrote. I'm sorry, but in this case still has a bug right? Because if you write `True/1`, the return is `False`, and `False` is not a echo from the value that was wrote. The same if you write `0`, the return is `True` instead `False`. I think that the logical is inverted. If I missing some understand, please, tell me what.

brainelectronics, 2022-11-12:
I'll investigate and debug this issue after #28 is merged and I have some working setup here. I still believe it is a bug, like you do. Let's see

brainelectronics, 2022-11-19:
Should be fixed as well, see https://github.com/brainelectronics/micropython-modbus/commit/fda9c10b14c060a94204ffd49a57476576524ed9 and #22

brainelectronics, 2022-12-11:
@beyonlo I think this bug is fixed and passing the test https://github.com/brainelectronics/micropython-modbus/blob/eee18c25162200d6c7120f7e5247ee44505aa8b1/tests/test_tcp_example.py#L498-L507

closed 2022-12-11 by brainelectronics

beyonlo, 2022-12-11:
Hi @brainelectronics 

> @beyonlo I think this bug is fixed and passing the test

I don't think so.

Are you tested the `2.0.0-rc9.dev33` version? I tested it now with the `2.0.0-rc9.dev33`. The `output_value=1` response is `OK,` but the `output_value=0` response is `NOT OK`. And now this version has another bug: the `output_value=1` do not set to 1 (True), the COIL still False. Details below:

**ModBus TCP Slave:**

```
register_definitions = {
    "COILS": {
        "RESET_REGISTER_DATA_COIL": {
            "register": 42,
            "len": 1,
            "val": 0
...
```

```
$ mpremote run tcp_client_example.py
Waiting for WiFi connection...
Connected to WiFi.
('192.168.1.3', '255.255.255.0', '192.168.1.1', '192.168.1.1')
Setting up registers ...
Register setup done
Running ModBus version: 2.0.0-rc9.dev33
Serving as TCP client on 192.168.1.3:502
Resetting register data to default values ...
Default values restored
```

**ModBus TCP Master:**

```
>>> from umodbus.tcp import TCP as ModbusTCPMaster
>>> host = ModbusTCPMaster(slave_ip='192.168.1.3', slave_port=502, timeout=5)
>>> from umodbus import version
>>> version.__version__
'2.0.0-rc9.dev33'
>>> host.read_coils(slave_addr=10, starting_addr=42, coil_qty=1)
[False]
>>> host.write_single_coil(slave_addr=10, output_address=42, output_value=1)
True <--- Response is OK, but do not write that 1 - see next command
>>> host.read_coils(slave_addr=10, starting_addr=42, coil_qty=1)
[False] <--- Still False, was not set 1
>>> host.write_single_coil(slave_addr=10, output_address=42, output_value=0)
True <--- Response is NOT OK. Should be False.
>>> host.read_coils(slave_addr=10, starting_addr=42, coil_qty=1)
[False]
```

brainelectronics, 2022-12-12:
Using the coil 42 might be not the best one to verify it 😄 

Looking at the TCP client example output you see `Default values restored`

Looking at the tcp client example code https://github.com/brainelectronics/micropython-modbus/blob/eee18c25162200d6c7120f7e5247ee44505aa8b1/examples/tcp_client_example.py#L137, the Register setup function is called in order to restore the default values of all registers. I use this to perform a "cleanup" after each test, check the `tearDown` function of the TCP test https://github.com/brainelectronics/micropython-modbus/blob/eee18c25162200d6c7120f7e5247ee44505aa8b1/tests/test_tcp_example.py#L627

The initial state of coil `42` is FALSE. It would be true for a very short time (after setting it and before resetting it)

If you test it with another coil the bug should be fixed

brainelectronics, 2022-12-13:
I think there might be a little misunderstanding. The function return value of `write_single_coil` and other setters is not depending on the set value. It is returning the result of the operation. So whether the operation of setting the specified value has been successfully performed (`True`) or not (`False`). In case of `False` also an additional error could raise. 

Getter functions like `read_coils` instead return the current state of the register(s) which is of course depending on a previously set value or external event. 

I'll further update the documentation to make it clear, contributes to #27

beyonlo, 2022-12-13:
Hi @brainelectronics 

>So whether the operation of setting the specified value has been successfully performed (True) or not (False). In case of False also an additional error could raise.

All right! So, the tests passed to me!

Thank you for one more bug fix :)