brainelectronics/micropython-modbus #22: Bug: Error when trying to write multiple coils (host.write_multiple_coils())

beyonlo, 2022-11-11:
This bug was extracted from PR #10 report, in this [reply](https://github.com/brainelectronics/micropython-modbus/pull/10#issuecomment-1189078241):

Error when trying to write multiple coils (`host.write_multiple_coils()`)

I not found examples on the docs about how is the correct way to write multiple coils, so I get a example from [here](https://docs.pycom.io/tutorials/hardware/modbus/)

```
>>> host.read_coils(slave_addr=10, starting_addr=123, coil_qty=1)
[True, False, False, False, False, False, False, False]
>>>
>>> host.write_multiple_coils(slave_addr=10, starting_address=123, output_values=[1,1,1,0,0,1,1,1,0,0,1,1,1])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "umodbus/serial.py", line 274, in write_multiple_coils
  File "umodbus/serial.py", line 173, in _send_receive
  File "umodbus/serial.py", line 190, in _validate_resp_hdr
ValueError: slave returned exception code: 1
>>> 
>>> host.write_multiple_coils(slave_addr=10, starting_address=123, output_values=[1,1,1,0,0,1])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "umodbus/serial.py", line 274, in write_multiple_coils
  File "umodbus/serial.py", line 173, in _send_receive
  File "umodbus/serial.py", line 190, in _validate_resp_hdr
ValueError: slave returned exception code: 1
>>> 
```

brainelectronics, 2022-11-12:
See [Modbus Application Protocol](https://modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf) section 6.11

> The request  data  contents  are  two  bytes:  CD  01  hex  (1100  1101  0000  0001  binary).  The binary bits correspond to the outputs in the following way:
```
Bit:     1 1 0 0 1 1 0 1 0 0 0 0 0 0 0 1
Output: 27 26 25 24 23 22 21 20 – – – – – – 29 28
```

brainelectronics, 2022-11-19:
@beyonlo could you try my fixes of https://github.com/brainelectronics/micropython-modbus/tree/feature/create-test-framework, this bug should now be fixed. At least this is what my new unittests report

beyonlo, 2022-11-19:
@brainelectronics I can't to test the because this [new version](https://github.com/brainelectronics/micropython-modbus/tree/feature/create-test-framework) has problem to connect:

**ModBus TCP Master  - ESP32_02:**

```Python
>>> from umodbus.tcp import TCP as ModbusTCPMaster
>>> host = ModbusTCPMaster(slave_ip='192.168.1.4', slave_port=502, timeout=5)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "umodbus/tcp.py", line 30, in __init__
OSError: [Errno 118] EHOSTUNREACH
>>> 
```

**ModBus TCP Slave  - ESP32_01:**
$ mpremote run tcp_client_example.py
Waiting for WiFi connection...
Waiting for WiFi connection...
Waiting for WiFi connection...
Connected to WiFi.
('192.168.1.4', '255.255.255.0', '192.168.1.1', '192.168.1.1')
Setting up registers ...
Register setup done
Serving as TCP client on 192.168.1.4:502

**Note:** I tried to use a old version (1.1.0) on for the Master and I have the same error. However I tried to connect via `MODSCAN32` (ModBus TCP Master for windows) and connected with success.

beyonlo, 2022-11-19:
In both, Master and Slave I installed via pip the version 1.2.0rc5.dev28:

```Bash
>> import upip
>>> upip.index_urls = ['https://test.pypi.org/pypi']
>>> upip.install('micropython-modbus')
Installing to: /lib/
Warning: test.pypi.org SSL certificate is not validated
Installing micropython-modbus 1.2.0rc5.dev28 from https://test-files.pythonhosted.org/packages/5d/75/753871c80d40e9d6659e6e7185b0022d26f137265c2c333ad0865079fb89/micropython-modbus-1.2.0rc5.dev28.tar.gz
```

I tried as well, just for sure, download the .zip from that [new version](https://github.com/brainelectronics/micropython-modbus/tree/feature/create-test-framework) and test it, and I have the same error.

beyonlo, 2022-11-22:
Hey @brainelectronics 

I tested the `1.2.0rc5.dev28` using `ModBus RTU` (where do not need `TCP` connection of course), but bug still the same:

**ModBus RTU Slave:**
For the slave I used this [rtu_client_example.py](https://github.com/brainelectronics/micropython-modbus/blob/feature/create-test-framework/examples/rtu_client_example.py)
`$ mpremote run rtu_client_example.py
`

**ModBus RTU Master:**

```Bash
>>> from umodbus.serial import Serial as ModbusRTUMaster
>>> rtu_pins = (18, 17)
>>> host = ModbusRTUMaster(baudrate=9600, data_bits=8, stop_bits=1, parity=None, pins=rtu_pins)
>>>
>>> host.read_coils(slave_addr=10, starting_addr=123, coil_qty=1)
[True, False, False, False, False, False, False, False]
>>>
>>> host.write_multiple_coils(slave_addr=10, starting_address=123, output_values=[1,1,1,0,0,1])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/umodbus/serial.py", line 274, in write_multiple_coils
  File "/lib/umodbus/serial.py", line 173, in _send_receive
  File "/lib/umodbus/serial.py", line 190, in _validate_resp_hdr
ValueError: slave returned exception code: 1
>>> host.write_multiple_coils(slave_addr=10, starting_address=123, output_values=[1,1,1,0,0,1,1,1])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/umodbus/serial.py", line 274, in write_multiple_coils
  File "/lib/umodbus/serial.py", line 173, in _send_receive
  File "/lib/umodbus/serial.py", line 190, in _validate_resp_hdr
ValueError: slave returned exception code: 1
>>>
>>> host.write_single_coil(slave_addr=10, output_address=123, output_value=0)
True
>>> host.read_coils(slave_addr=10, starting_addr=123, coil_qty=1)
[False, False, False, False, False, False, False, False]
```

**Scenario used:**
ESP32-S3_01 (ModBus 1.2.0rc5.dev28) --------  UART ------------- ESP32-S3_02 (ModBus 1.2.0rc5.dev28)

**Installation method: via pip:**
```Bash
>>> import upip
>>> upip.index_urls = ['https://test.pypi.org/pypi']
>>> upip.install('micropython-modbus')
Installing to: /lib/
Warning: test.pypi.org SSL certificate is not validated
Installing micropython-modbus 1.2.0rc5.dev28 from https://test-files.pythonhosted.org/packages/5d/75/753871c80d40e9d6659e6e7185b0022d26f137265c2c333ad0865079fb89/micropython-modbus-1.2.0rc5.dev28.tar.gz
```

brainelectronics, 2022-12-17:
Relates to #24

brainelectronics, 2022-12-17:
Relates to #38

commit 0cbb4444 (2022-12-27): add test to verify write multiple coils fix #22, #38 is still open and valid
commit 16869a44 (2022-12-27): fix writing multiple coils in TCP, resolve #22
closed 2022-12-27 by brainelectronics