brainelectronics/micropython-modbus #23: Bug: error when trying to write multiple registers (`host.write_multiple_registers()`)

beyonlo, 2022-11-11:
This bug was extracted from PR #10 report, in this [reply](https://github.com/brainelectronics/micropython-modbus/pull/10#issuecomment-1189078241):

Error when trying to write multiple registers (`host.write_multiple_registers()`)

```
>>> host.write_single_register(slave_addr=10, register_address=93, register_value=44, signed=False)
True
>>> host.read_holding_registers(slave_addr=10, starting_addr=93, register_qty=1, signed=False)
(44,)
>>> 
>>> host.write_multiple_registers(slave_addr=10, starting_address=93, register_values=[2, -4, 6, -256, 1024], signed=True)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "umodbus/serial.py", line 291, in write_multiple_registers
  File "umodbus/serial.py", line 173, in _send_receive
  File "umodbus/serial.py", line 190, in _validate_resp_hdr
ValueError: slave returned exception code: 1
>>> 
>>> host.write_multiple_registers(slave_addr=10, starting_address=93, register_values=[2, 1024], signed=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "umodbus/serial.py", line 291, in write_multiple_registers
  File "umodbus/serial.py", line 173, in _send_receive
  File "umodbus/serial.py", line 190, in _validate_resp_hdr
ValueError: slave returned exception code: 1
>>> 

```

brainelectronics, 2022-11-12:
See [Modbus Application Protocol](https://modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf) section 6.12

brainelectronics, 2022-12-16:
might be connected with this line
https://github.com/brainelectronics/micropython-modbus/blob/eee18c25162200d6c7120f7e5247ee44505aa8b1/umodbus/tcp.py#L171
due to 

- always using `signed=False`
- always using only first element of unpacked data

commit 9b0662bd (2022-12-27): fix writing multiple registers in TCP, resolve #23
closed 2022-12-27 by brainelectronics