brainelectronics/micropython-modbus #26: Docs: request a simple doc about to create/extend `register definitions` (rules and meanings)

beyonlo, 2022-11-11:
In my opinion, is very important, to prevent errors, to have a simple doc to explain two things:

1. What is rules on the `register definitions`, I mean, what and how each argument need to be constructed.
2.  What means (what it will affect) each argument. Example: until now is not very clear to me what means the `len`, like as `'len': 1`. The reason that is not `very` clear yet is that some times I changed the `len` to other value, and nothing change. Or because my `register definitions` is not correct - so, is a one more good reason to have a doc explaining in details about `register definitions`.

brainelectronics, 2022-11-12:
found by #10

brainelectronics, 2022-12-08:
Relates to #31

commit 9f967ae5 (2022-12-08): update docs USAGE, resolves #26
closed 2022-12-11 by brainelectronics