brainelectronics/micropython-modbus #31: It would be nice if there was an example of ModBus Slave talking to real hardware

jd-boyd, 2022-11-29:
This could be a GPIO or I2C device.

I'm guessing that the correct approach is to inherit `ModbusRTU` (or `ModbusTCP`)  and then override functions like `set_ist`, `set_coil`, `get_coil`, etc.

beyonlo, 2022-11-30:
Hi @jd-boyd 

Yes, that will be nice. 

Currently, how are you doing to get/set on the real hardware from the Slave, from/to the `register_definitions?` I'm still do not have reached this part, I still doing tests with the `register_definitions,` like [this example](https://github.com/brainelectronics/micropython-modbus/blob/develop/examples/set-example.json), without to integrate with real hardware. I imagine that the best solution for now is a loop reading/writing from real hardware and updating/checking the `register_definitions.` As `register_definitions` is a dict, always that loop find a different value on the hardware, update the `register_definitions` dict, that will be what `Slave` will read. The same for write new value on the hardware, loop check if are there a different value on the `register_definitions` (that was set by Master), and write on the hardware. I really do not know what is the best solution.

brainelectronics, 2022-11-30:
Big thank you to @beyonlo. You are the best support! 

@jd-boyd I think you want something like using a coil as a GPIO, whenever it is physically modified the register value should be updated and if it modified by a Modbus Host, it's physical state should change. I might implement something like that on the weekend

commit d8fce045 (2023-01-06): add on_set and on_get callbacks to modbus, resolves #31
closed 2023-01-06 by brainelectronics