brainelectronics/micropython-modbus #37: Reading holding registers does always return all registers

brainelectronics, 2022-12-09:
Related to #25, similar to #36, contributes to #35, found during https://github.com/brainelectronics/micropython-modbus/pull/33#issuecomment-1336274272 section "Problem 4" 

Defining a holding register with more than one element, but requesting only 1 element, returns all elements of the register.

```
register_definitions = {
    "HREGS": {
        "ANOTHER_EXAMPLE_HREG": {
            "register": 94,
            "len": 9,
            "val": [29, 38, 0, 1600, 2150, 5067, 2564, 8450, 3456],
            "description": "Example HREGS registers with length of 9, Hregs (setter+getter) [0, 65535]",
            "range": "",
            "unit": ""
        }
    }
}
```

```
>>> host.read_holding_registers(slave_addr=10, starting_addr=94, register_qty=1, signed=False)
(29, 38, 0, 1600, 2150, 5067, 2564, 8450, 3456)
# expectation would be (29,)
```

**Reason/root cause:**
The response quantity is defined as the length of the returned byte list, see https://github.com/brainelectronics/micropython-modbus/blob/2f1e307fa087e2f97fa6031a4bfb5b8d8bdddcd2/umodbus/functions.py#L379-L380 not taking into account the actual amount of requested data. Thereby the format to unpack the data is `>hhhhhhhhh` instead of `>h` as the response byte array lenght is 9, but the register quantity of interest is 1

brainelectronics, 2023-01-08:
@beyonlo I think this is also solved with #51 and release 2.3.1

beyonlo, 2023-01-08:
> @beyonlo I think this is also solved with #53 and release 2.3.1

@brainelectronics exactly, this issue is solved!

**Master:**

```Python
>>> from umodbus import version
>>> print('Running ModBus version: {}'.format(version.__version__))
Running ModBus version: 2.3.1-rc26.dev53
>>> host.read_holding_registers(slave_addr=10, starting_addr=93, register_qty=3, signed=False)
response var from serial.py file is: bytearray(b'\n\x03\x06\x00\x1d\x00&\x00\x00_\x8c')
(29, 38, 0)
>>> host.read_holding_registers(slave_addr=10, starting_addr=93, register_qty=5, signed=False)
response var from serial.py file is: bytearray(b'\n\x03\n\x00\x1d\x00&\x00\x00\x06@\x08fj\x19')
(29, 38, 0, 1600, 2150)
>>> host.read_holding_registers(slave_addr=10, starting_addr=93, register_qty=1, signed=False)
response var from serial.py file is: bytearray(b'\n\x03\x02\x00\x1d\xdd\x8c')
(29,)
>>> 

```

closed 2023-01-09 by brainelectronics