brainelectronics/micropython-modbus #43: Serial problem on RPi Pico W

animalmutch, 2022-12-29:
Issue:

When creating a modbus master on RPi Pico W, you get the following error:

```>>> m = ModbusRTU(addr=1, pins=(0,1))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "umodbus/serial.py", line 58, in __init__
  File "umodbus/serial.py", line 104, in __init__
ValueError: expecting a Pin```

If you create Pin objects and use these instead, you get the following error:

```
>>> tx_pin = Pin(0)
>>> rx_pin = Pin(1)
>>> m = ModbusRTU(addr=1, pins=(tx_pin, rx_pin))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "umodbus/serial.py", line 58, in __init__
  File "umodbus/serial.py", line 104, in __init__
ValueError: bad TX pin
```

beyonlo, 2022-12-29:
Hello @animalmutch 

1. What version of this lib are you using?
2. Could you please paste the complete code, how are you importing the modules?
3. MicroPython version that you are using is as well a good information.

In the new versions works fine to me:

```Bash
>>> from umodbus.serial import Serial as ModbusRTUMaster
>>> rtu_pins = (17, 18)
>>> host = ModbusRTUMaster(baudrate=115200, data_bits=8, stop_bits=1, parity=None, pins=rtu_pins, ctrl_pin=15)
>>>
```

animalmutch, 2022-12-29:
Thanks for your swift reply @beyonlo .

I am using version 2.1.1 of the lib (seems to be the latest release). I am not installing using upip, because upip doesn't seem to come bundled with micropython any more (they use mip instead now).

I am using the latest UF2 from raspberrypi.com (https://micropython.org/download/rp2-pico-w/rp2-pico-w-latest.uf2):

`MicroPython v1.19.1-782-g699477d12 on 2022-12-20; Raspberry Pi Pico W with RP2040`

Using the exact same code as you above, I get the following:

[image]

Seems most likely there's an issue with using the lib without installation. Can you advise any further on installation on RPi pico?

brainelectronics, 2022-12-29:
Relates to #17

brainelectronics, 2022-12-29:
@animalmutch according to the [MicroPython v1.19.1 RP2 UART docs](https://docs.micropython.org/en/v1.19.1/rp2/quickref.html#uart-serial-bus) the RX and TX pins shall be specified as `Pin` objects, which is different than in the e.g. [ESP32 UART setup](https://docs.micropython.org/en/v1.19.1/esp32/quickref.html#uart-serial-bus) where A) any pin can be used, and B) a pin as integer shall be used.

The following example should work for you on a RP2

```python
>>> from umodbus.serial import Serial as ModbusRTUMaster
>>> rtu_pins = (Pin(4), Pin(5))   # TX, RX
>>> host = ModbusRTUMaster(baudrate=115200, data_bits=8, stop_bits=1, parity=None, pins=rtu_pins, ctrl_pin=15)
>>>
```

Additionally the UART pins on RP2 can only be GPIO 4/5 or 8/9 for UART1 according to the docs

> There are two UARTs, UART0 and UART1. UART0 can be mapped to GPIO 0/1, 12/13 and 16/17, and UART1 to GPIO 4/5 and 8/9.

which explains your

> ```
> >>> tx_pin = Pin(0)
> >>> rx_pin = Pin(1)
> >>> m = ModbusRTU(addr=1, pins=(tx_pin, rx_pin))
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "umodbus/serial.py", line 58, in __init__
>   File "umodbus/serial.py", line 104, in __init__
> ValueError: bad TX pin
> ```

But there is also a not yet implemented feature in this lib (never call it bug 🤦 ), see https://github.com/brainelectronics/micropython-modbus/blob/469968484a13d4ee354175ad04b22a768a6a48f9/umodbus/serial.py#L58 where the `uart_id` is always set to `1` without the possibility to change it on ModbusRTU class level. So I assume the UART0 can not be used with pins GPIO 0/1, 12/13 and 16/17

animalmutch, 2022-12-29:
Thanks so much @brainelectronics ! I get it now, and it's working. It was just a mismatch between the pins I was using and the UART ID.

For reference, the following works perfectly:

```
modbus_host = ModbusRTUMaster(
            uart_id=0,
            baudrate=115200,
            pins=(Pin(0), Pin(1))
        )
```

For some reason Pin(4) and Pin(5) with uart_id=1 doesn't work (well, it doesn't cause a problem with your lib, but it physically doesn't work still, so perhaps a hardware issue).

Anyway, thanks again!

brainelectronics, 2022-12-31:
Resolved by #45

closed 2022-12-31 by brainelectronics