brainelectronics/micropython-modbus #54: Instructions for install on boards without network capabilities

toddkarin, 2023-01-07:
Could you please add instructions for boards without the ability to install using mip?

brainelectronics, 2023-01-07:
Hey @toddkarin have you tried to install the lib as documented here? 

[ReadTheDocs MicroPython Modbus Setup Installation without network connection](https://micropython-modbus.readthedocs.io/en/latest/SETUP.html#without-network-connection)

Simply copy the `umodbus` folder to the pyboards sub folder `lib/umodbus`

brainelectronics, 2023-01-29:
@toddkarin is your request solved? If so could you please close this issue, otherwise leave a comment.

closed 2023-07-05 by brainelectronics

tom9672, 2023-07-25:
I test on boards without network capabilitie, by copy the umodbus folder to the pyboards sub folder `lib/umodbus`.
Works well, as same as mip install.