brainelectronics/micropython-modbus #60: Cannot run example code (RTU Slave)

MGabeler, 2023-02-02:
### Description

Hi,

I installed this library on my RP Pi Pico (H), micropython. I get an syntax error running the example code from the library (Slave RTU). The error exists in the serial.py sub code of the library line 2 of the example code : from umodbus.serial import ModbusRTU(Modbus). When I run serial.py I got an error about the costum packages. Do I miss any extra necessary libraries that are used next to the Micropython-modbus library? I did read other questions about running this library on a RP Pi Pico but it didn't answer my question / I didn't help me fix this problem.

### Reproduction steps

This is the error I got when running the example code Slave RTU.

Traceback (most recent call last):
  File "<stdin>", line 2
SyntaxError: invalid syntax

this is the code I'm running:

from machine import Pin
from umodbus.serial import ModbusRTU(Modbus)

# the following definition is for a RP2
rtu_pins = (Pin(0), Pin(1))     # (TX, RX)
uart_id = 0

slave_addr = 1             # address on bus as client

client = ModbusRTU(
    addr=slave_addr,        # address on bus
    pins=rtu_pins,          # given as tuple (TX, RX)
    baudrate=9600,        # optional, default 9600
    data_bits=8,          # optional, default 8
    stop_bits=1,          # optional, default 1
    parity=None,          # optional, default None
    ctrl_pin=18,          # optional, control DE/RE
    uart_id=uart_id         # optional, default 1, see port specific documentation
)

register_definitions = {
    "COILS": {
        "EXAMPLE_COIL": {
            "register": 123,
            "len": 1,
            "val": 1
        }
    },
    "HREGS": {
        "EXAMPLE_HREG": {
            "register": 93,
            "len": 1,
            "val": 19
        }
    },
    "ISTS": {
        "EXAMPLE_ISTS": {
            "register": 67,
            "len": 1,
            "val": 0
        }
    },
    "IREGS": {
        "EXAMPLE_IREG": {
            "register": 10,
            "len": 1,
            "val": 60001
        
    }
}}

### MicroPython version

v.1.19.1

### MicroPython board

Raspberry Pico

beyonlo, 2023-02-02:
Hello @MGabeler 

> I installed this library on my RP Pi Pico (H), micropython. I get an syntax error running the example code from the library (Slave RTU). 

I believe that you are talking about this [tcp_client_example.py](https://github.com/brainelectronics/micropython-modbus/blob/develop/examples/tcp_client_example.py) right?

>The error exists in the serial.py sub code of the library line 2 of the example code : from umodbus.serial import ModbusRTU(Modbus). 

Look that (conforming [tcp_client_example.py](https://github.com/brainelectronics/micropython-modbus/blob/develop/examples/tcp_client_example.py)), [line 20](https://github.com/brainelectronics/micropython-modbus/blob/develop/examples/tcp_client_example.py#L20), the correct is import modbus client classes is `from umodbus.tcp import ModbusTCP`. You are using `from umodbus.serial import ModbusRTU(Modbus)` Maybe this can be the problem.

A important aditional information to paste here is: what ModBus version of this library that you are using:

```Python
from umodbus import version
print('Used micropthon-modbus version: {}'.format(version.__version__))
```

Another very useful information, if you can, is to paste the real output running on terminal.

MGabeler, 2023-02-02:
I have this library: Used micropthon-modbus version: 2.3.3
I use Modbus RTU with Tx and Rx on 0 and 1 to a max485 module to convert to RS485. wait, now it works, I just found the error. instead of using: from umodbus.serial import ModbusRTU(Modbus)
you need to use this: from umodbus.serial import ModbusRTU
Now I'm going to set the Pi 4 B as a master. that's the next step. I have no errors on the slave code right now. so will be back when the master doesn't work xD Can I better use Python instead of micropython for the master? (Pi 4B)
and thnx for the fast response :)

beyonlo, 2023-02-02:
>wait, now it works, I just found the error. instead of using: from umodbus.serial import ModbusRTU(Modbus) you need to use this: from umodbus.serial import ModbusRTU 

Yes, that was exactly what I wrote for you above!

>Now I'm going to set the Pi 4 B as a master. that's the next step. I have no errors on the slave code right now. so will be back when the master doesn't work xD Can I better use Python instead of micropython for the master?

This ModBus library works on MicroPython, and there is a [MicroPython version for Unix](https://github.com/micropython/micropython/tree/master/ports/unix) that run on Linux (including ARM arch). The Pi 4 B is ARM, so you can install tha MicroPython on PI4 (running linux) too. So, my suggestion is you just install MicroPython on the PI4, and use the ModBus Master of this Lib!

>thnx for the fast response :)

No problem, you are wellcome!

brainelectronics, 2023-02-22:
Hi @MGabeler 
is your issue solved with the help of @beyonlo answer? If so, could you close this issue? :)

MGabeler, 2023-02-22:
Hi, that would indeed solve my problem i suppose, but unfortunately i cant find how to install micropython on a raspi 4. I can not find a raspi 4 version on micropython.org

beyonlo, 2023-02-23:
Hello @MGabeler 

> Hi, that would indeed solve my problem i suppose, but unfortunately i cant find how to install micropython on a raspi 4. I can not find a raspi 4 version on micropython.org

There is no ready to go `MicroPython` binary to download and use on `RP4,` but, logged in your `RP4` (with Linux), you can just to download the `Micropython` source code and compile by your self. That `url` above that I pasted for you has the complete instructions to compile, including dependencies.

brainelectronics, 2023-02-23:
If you really want to use MicroPython on a Linux (Raspberry) you could go with a Docker solution. I use this to test this lib in the CI Chain. Further details are explained in the [MicroPython Modbus testing chapter](https://micropython-modbus.readthedocs.io/en/latest/TESTING.html)

brainelectronics, 2023-07-05:
@MGabeler I'll close this issue for now. In case you face some other issue please open a new issue

closed 2023-07-05 by brainelectronics