brainelectronics/micropython-modbus #63: Cite this repository

GimmickNG, 2023-03-19:
### Description

I would like to cite this repository. Github provides a means of doing so via the CITATIONS file (IIRC). Would this be supported, or should I just cite the start page?

### Reproduction steps

N/A

### MicroPython version

N/A

### MicroPython board

other

brainelectronics, 2023-03-20:
Hi @GimmickNG 
Great proposal, I would support this of course. May you can create a PR for that? It should get in way faster than the other PR 😅 as there as no functional changes. 

The very initial author has been "Pycom Limited" some time ago as mentioned in the header. 
Since then you, wthomson, see #32, and me contributed. 

Also beyonlo has to be mentioned, he's the best tester of this package since the beginning.

GimmickNG, 2023-03-20:
@brainelectronics I'm not sure the etiquette of citing public repositories - for the Pytorch Lightning repository, it seems to be "William Falcon and the Pytorch Lightning team", for PyTorch it's all the authors listed in the paper for PyTorch rather than all the contributors.

(https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/about-citation-files - the reference for the CITATIONS file)

By the way, thanks for offering to include me in the citations, but since I'm citing the work I'm not sure it would be appropriate to retroactively add myself. Since you are the owner of the repository, and since this has been forked from the original, I suggest the authors could be "Pycom Limited", then you and "the MicroPython Modbus team", like with PyTorch Lightning.

Of course, since this risks erasing others' contributions, I'm interested in hearing their opinion before making these changes. @beyonlo @wthomson

GimmickNG, 2023-03-23:
@beyonlo No worries! :) The "Micropython Modbus team" is just my suggestion as shorthand to refer to the contributors of this repository who may also work on this library in the future, since there may be others who also contribute. The reason I avoided including myself, you or @wthomson is that the CITATION file is (afaik) not generally supposed to change, so once a citation is created that's more or less how it'll be used to refer to it in the future. By including people apart from the original author and the current owner of the repository (i.e @brainelectronics) it favours people who appeared _first_. That being said, I'm probably overthinking this, so I could modify it to include the names of everyone who has worked thus far on this.

And yes, it would be a good idea to include contributors in release notes - and that's a more flexible option for those who end up visiting the repository.

beyonlo, 2023-03-24:
@GimmickNG I'm sorry, maybe my comment was not clear. In my opinion you are right, to have in the `CITATION` just the @brainelectronics and the **The Micropython Modbus team**, where **The Micropython Modbus team** can be everyone that already helped with this lib, past, present and future - that's is very good! My mention about the "although it is not known who is part of the team" is that is a fact, and not that I deserve or that I want to be included in the `CITATION,` no - apologies if I passed that feeling to you :) But, a good way in my opinion to show who is part of team is to do that suggestion about each release notes to mentioning who helped :) This is just a suggestion with my experience with `MicroPython release notes`, that I was remembered when you requested my opinion about this subject!

brainelectronics, 2023-03-24:
I'll try to take care of that in the upcoming 2.4.0 release. As all of this is automated, I'll manually edit the release text

closed 2023-03-24 by brainelectronics