brainelectronics/micropython-modbus #66: Raspberry Pico Modbus TCP IO Device

cprobotik, 2023-03-22:
### Description

Iam New with micropython and the Raspberry Pico W and i would like to create a Modbus TCP IO Divice for a PLC on an Raspberry Pico W
I would read some GPIO to send they over Modbus TCP to the Modbus Master

Can someone show me an code example to fix my problem.

Thank you so much

### Reproduction steps

1.install Micropython an Modbus on Raspberry was okay
2.I use the Thonny  IDE and I can run the TCP Client Example on the Raspberry Pico
3. but I down know how can I send GPIO Data to the Modbus Master
4. I need a example for send A Bool GPIO value (coil)
5. I need a example for send A ADC value (holding register)

### MicroPython version

v1.19.1

### MicroPython board

Raspberry Pico

### MicroPython Modbus version

```python
v2.3.3
```

beyonlo, 2023-03-24:
@cprobotik [Here](https://github.com/brainelectronics/micropython-modbus/tree/develop/examples) there are all examples. As you `PICO-W` will be the `Modbus TCP Slave`, you can to follow this example: [tcp_client_example.py](https://github.com/brainelectronics/micropython-modbus/blob/develop/examples/tcp_client_example.py)

About to serve a GPIO and/or an ADC on the Slave to the Master request it, you can use the amazing callback feature. In that example above, there is a function `def my_inputs_register_get_cb(reg_type, address, val):` that does exactly what you want. And doc about callback feature you can found [here](https://micropython-modbus--51.org.readthedocs.build/en/51/USAGE.html#optional-callbacks)

cprobotik, 2023-03-25:
Thank you for your answer, the callback Funktion works great.

The examples you have given are perfekt.

Thank you for the Professional Code.

brainelectronics, 2023-04-04:
@cprobotik may you close this issue if your request has been resolved?

closed 2023-04-04 by cprobotik