brainelectronics/micropython-modbus #72: Control Pin problem (remained to HIGH after several time of communication)

panther50500, 2023-04-25:
### Description

Hello,
I use it with a Raspberry PICO with modbus rtu protocol and RTS control PIN activated to manage a MAX485 device.
So, I had to modify few lines in the serial.py file because after a few time of communication, the control PIN remained to HIGH and blocked the MAX485 in transmit mode. I think the problem arrives when an overflow of time.ticks_us() occurs.

Here my modifications : 

line 110 ( to inialise RTS pin to 0 at reboot, in case of freeze or execution program stopped when control PIN activated, else it stayed to HIGH and block reception) :

```        
if ctrl_pin is not None:
    self._ctrlPin = Pin(ctrl_pin, mode=Pin.OUT)
    self._ctrlPin(0)
else:
    self._ctrlPin = None
```

line 260 (use time.ticks_diff() ) :
```
if self._ctrlPin:
    total_frame_time_us = self._t1char * len(serial_pdu)
    #while time.ticks_us() <= send_start_time + total_frame_time_us:
    while time.ticks_diff(time.ticks_us(), send_start_time) < total_frame_time_us :
        #machine.idle()
        time.sleep_us(20)   # arbitrary time
    self._ctrlPin(0)

```

### Reproduction steps

1. start continous communication at 19200bauds
2. wait several minutes
3. Communication interrupted because ctrl PIN stay to HIGH
...

### MicroPython version

v1.9.1

### MicroPython board

Raspberry Pico

### MicroPython Modbus version

```python
# e.g. v2.3.3
# use the following command to get the used version
import os
from umodbus import version
print('MicroPython infos:', os.uname())
print('Used micropthon-modbus version:', version.__version__))
```

### Relevant log output

```bash
MicroPython infos: (sysname='rp2', nodename='rp2', release='1.19.1', version='v1.19.1-910-g4937174b4 on 2023-03-01 (GNU 12.1.0 MinSizeRel)', machine='Raspberry Pi Pico with RP2040')
Used micropthon-modbus version: 2.3.3
```

### User code

```python
#!/usr/bin/env python
#
# Copyright (c) 2019, Pycom Limited.
#
[... 402 lines cut ...]

        if len(req) < 8:
            return None

        if req[0] not in unit_addr_list:
            return None

        req_crc = req[-Const.CRC_LENGTH:]
        req_no_crc = req[:-Const.CRC_LENGTH]
        expected_crc = self._calculate_crc16(req_no_crc)

        if (req_crc[0] != expected_crc[0]) or (req_crc[1] != expected_crc[1]):
            return None

        try:
            request = Request(interface=self, data=req_no_crc)
        except ModbusException as e:
            self.send_exception_response(
                slave_addr=req[0],
                function_code=e.function_code,
                exception_code=e.exception_code)
            return None

        return request
```

panther50500, 2023-07-10:
> Hey @panther50500 could you retest the reported issue with 2.3.5, it should have been fixed by #75

Hello. I'm sorry I couldn't rewire and test before today. On the raspberry pico, I made a modification line 280 to add "time.sleep_us(self._t1char)" otherwise I have errors and on the scope, the data is covered by the rts signal : 

      if self._has_uart_flush:
          self._uart.flush()
          **time.sleep_us(self._t1char)**
      else:
          sleep_time_us = (
              self._t1char * len(modbus_adu) -    # total frame time in us
              time.ticks_diff(send_finish_time, send_start_time) +
              100     # only required at baudrates above 57600, but hey 100us
          )
          time.sleep_us(sleep_time_us)

For the moment, I have tested at 19200 & 115200 with and whithout "if self._has_uart_flush" condition and until now, I have no errors.

brainelectronics, 2023-07-19:
Hi again @panther50500 could you retest the reported issue with 2.3.7, it should have been fixed by https://github.com/brainelectronics/micropython-modbus/pull/79, I've added you proposed wait time after the flush.

panther50500, 2023-07-25:
> Hi again @panther50500 could you retest the reported issue with 2.3.7, it should have been fixed by #79, I've added you proposed wait time after the flush.

Hi, I tested the 2.3.7 revision during few days and with different baudrates (9600, 19200, 115200 and 500000) also with and without the has_uart_flush condition. It seems to be good for the driving of the RTS signal like that. Thanks