brainelectronics/micropython-modbus #92: Does this require a specific micropython firmware version ?

Philj6321, 2024-05-30:
### Description

I've tried for most of the day yesterday to get this to work with a Pico W - any examples I can find seem to use an earlier version of Micropython (Micropython v1.19.1-1008-gc046b23ea)

No matter what I do I always get this error returned. (See log output)

### Reproduction steps

1.
2.
3.
...

### MicroPython version

v1.22.2

### MicroPython board

Raspberry Pico

### MicroPython Modbus version

```python
2.3.7
```

### Relevant log output

```bash
MPY: soft reboot
waiting for wifi connection...
waiting for wifi connection...
waiting for wifi connection...
Connected to WiFi, Pico W IP : 192.168.10.171
Traceback (most recent call last):
  File "<stdin>", line 64, in <module>
  File "/lib/umodbus/tcp.py", line 86, in __init__
OSError: [Errno 103] ECONNABORTED
```

### User code

```python
import time
from umodbus.tcp import TCP as ModbusTCPMaster
import network

# =============================================
# connect to a wifi network

station = network.WLAN(network.STA_IF)
if station.active() and station.isconnected():
    station.disconnect()
    time.sleep(1)
station.active(False)
time.sleep(1)
station.active(True)

station.connect("MY_SSID", "PASSWORD")
time.sleep(1)

while True:
    print('waiting for wifi connection...')
    if station.isconnected():
        print(f'Connected to WiFi, Pico W IP : {station.ifconfig()[0]}')
        break
    time.sleep(1)

# TCP Slave setup
port = 502
slave_addr = 1
ireg_address = 0
register_qty = 2

# IP Address of Modbus TCP Server
ip = '192.168.10.171'

# Setup Modbus TCP Client
modbus_client = ModbusTCPMaster(slave_ip=ip,slave_port=port,timeout=5)
print(f'Resquesting data from modbus TCP Server at {ip}:{port}')

register_value = modbus_client.read_input_registers(slave_addr, starting_addr=ireg_address, register_qty= register_qty)
print('Status of IREG from {} to {}'.format(ireg_address+register_qty, register_value))
```

Philj6321, 2024-06-03:
So apparently the answer is YES - I found a picoW, that just happened to have :- 
MicroPython v1.19.1-1016-gb525f1c9e on 2023-04-14; Raspberry Pi Pico W with RP2040 installed.
and it works as predicted - if someone knows where to get that firmware version from, please let me know, of-course if someone has gone the stage further and got this lib to work with 1.22 or higher, please let us know

algorni, 2024-11-07:
+1 using a raspi pico 2 for the RTU and getting: 

MPY: soft reboot
Traceback (most recent call last):
  File "<stdin>", line 5, in <module>
  File "/lib/umodbus/serial.py", line 19, in <module>
ImportError: no module named 'umodbus.functions'